newHeiko/wiFred #40: Esp32-S2 upgrade to esp32-s3

MOppHub, 2022-09-13:
The eps32-s2 modules are in the meantime NRND. So long term procurement could become difficult.
Maybe it's worth to do an upgrade to esp32-s3 modules, when the next PCB redesinge is necessary.

This would also enable to implement Bluetooth communication as alternative option in the future.

newHeiko, 2022-10-05:
With Espressif's longevity commitment https://www.espressif.com/en/products/longevity-commitment for the S2 reaching into 2031 and the S3 only a year longer, I hope there will be a different option when 2031 gets closer.

newHeiko, 2026-08-20:
As the ESP32-S2-WROOM and -WROVER are now essentially unavailable, a different module will be required.

Another project based on this will go into discussions with Espressif to decide on which module to use. This project will wait for the results of that discussion and then decide to either follow them or choose a different one.

DL2x, 2026-09-02:
I made a Version based on the ESP32-S3-MINI-1. It can be found here: https://forgejo.sbmc.dev/DL2x/wifred (alltough the code is not yet updated to new GPIO mapping)
It also adds a button for F9 to allow for a full keypad with 0 to 9. If you'd like me to, Ill move it over to github and create a pull request.

newHeiko, 2026-09-03:
Thanks for the offer, but I'm waiting for the other project's decision before I'll change the PCB to a new ESP chip.

Trying to reduce the number of versions in my direct surroundings ;)

Heiko

DL2x, 2026-09-03:
Yeah, thats understandable.
If the suggested chip comes out to the S3-MINI-1 or one with a similar landing pattern, feel free to use my design.
Same for any people searching for an Alternative until the decision is made.