newHeiko/wiFred #48: wiFRED use with command stations equipped with the WiThrottle protocol

HonzaM44, 2023-05-17:
It seems not very practical to have to use computer with JMRI for connection to the Train command station. There might be several ways how to use the wiFRED operating on WiThrottle protocol, either using the Digitrax  LNWI unit to connect it to the LocoNet used to standard FREDs, or to use some command station equipped with the wifi and WiThrottle protocol, such as YaMoRC YC7100 or YC7100, TCS-2. Probably as well the ESU Smart Control.
It is only necessary that it would exist a way of easy wifi connection to the LNWI or command stations mentioned above.

danielb987, 2023-05-17:
Running JMRI on a Raspberry Pi is a good choice. There is a prebuilt image for RPi with JMRI here: https://mstevetodd.com/rpi

And instructions to install JMRI on RPi if you want to do it yourself:
https://www.jmri.org/install/Raspbian.shtml

HonzaM44, 2023-05-17:
I want to use the wiFRED  without the need of PC with installed JMRI - WiThrottle might be used on the train command station which “understand” to it! Same way can communicate mobile phones using WiThrottle Application (iOS) or Engine Driver (Android).

newHeiko, 2023-05-17:
Well, the wiFred is designed under the premise that you have a smartphone, a tablet or a notebook computer to _configure_ it (WiFi and loco addresses at the minimum). You don't need JMRI for that, just WiFi and a web browser.

But once you've done that, sure, you can use it with DCC++EX or an LNWI (unless I broke something since I last tested) or with the YaMoRC WiFi https://youtu.be/azPPXs0PULk?t=197 or the upcoming Uhlenbrock WiFi thingie or Intellibox neo.

I have no idea about the ESU Smart Control - does that even support wiThrottle? - or the TCS command station. If I can get my hands on one, I might test it.

biteater, 2023-05-22:
You can find the documentation in the directory 'documentation' of this wiFred Github repository or here:
https://newheiko.github.io/wiFred/documentation/docu_en.html#throttle

I suggest creating separate issues for the additional topics you mentioned, just to keep the discussion focused.

newHeiko, 2023-05-22:
Let's pick this apart:

- I'm not aware of any place or person selling ready-made wiFreds. I personally do not and I will not.
- You can either leave out three of the four slide switches when assembling the wiFred (you'll need to solder a jumper wire in then for the three not in use) or just configure only one address and ignore the other three switches, IMHO. Or maybe I don't understand your "only one address" concern correctly.
- Easy setting of the address is something I will not implement, as it's a very explicit Fremo rule to not allow that. Have a look at one of the many other projects like https://model-railroad-hobbyist.com/sites/model-railroad-hobbyist.com/files/esp32_sma30_pcnry_v1_0_0.pdf / https://forum.mrhmag.com/post/sma30-a-simplified-wifi-throttle-you-can-customize-12211948 or https://github.com/flash62au/WiTcontroller (by JMRI's / EngineDriver's very own Peter Akers) and see if one of those better suits your use needs.
- The wiFred will already fail to pick up an address that's in use by another throttle (if the wiThrottle server supports it as JMRI does - from 5.4 onwards also for wiFred<->wiFred). I do need to add some LED blink codes to let the user know, though (see #45 )
- The wiFred enables "heartbeat" timeout if the wiThrottle server offers it. As JMRI does. So after the heartbeat timeout, JMRI will stop the loco (I don't know how other wiThrottle-commpatible servers handle that) But when the connection is gone, there's nothing else the wiFred can do to stop the loco.

Heiko

HonzaM44, 2023-05-25:
Hi,

you wrote, that for setting of wiFRED can be, beside the notebook with JMRI, used as well a smartphone - what application should be used for this purpose and how to do it?

As regards four slide switches, their use is in fact against the FREMO rules, as in this case, user can choose any one of the four Locos - user alone should not be able to control more than one loco with its throttle. 

Jan

newHeiko, 2023-05-25:
> you wrote, that for setting of wiFRED can be, beside the notebook with JMRI, used as well a smartphone - what application should be used for this purpose and how to do it?

You can use any web browser (Safari, FOSS browser, Chrome, Brave, Opera, Firefox, Chromium, Edge...). Connect your Smartphone or Tablet or Laptop to the wiFred-configXXXXXX network, wait for the captive portal to load, or - if it does not automatically - open it through the "Sign into network" button or just enter http://neverssl.com or any other non-https website into your browser.  

> As regards four slide switches, their use is in fact against the FREMO rules, as in this case, user can choose any one of the four Locos - user alone should not be able to control more than one loco with its throttle.

I read the Fremo rules more like the user should not be allowed to choose any loco address, but if your group implements that in a different way, don't install slide switches 2-4. Or just don't configure loco addresses for them.

Heiko

HonzaM44, 2023-05-26:
Hi,
thanks for your answers and explanations. There is one more item I would like to be clarified. According to FREMO rules, it must not be possible to connect simultaneously two throttles having the same address. If there would be two (wired) FREDs programmed with the same address, than after one is already connected and active on the LocoNet bus, when you connect the other one, the address in this second one is automatically deleted, if somebody wants to use this FRED again, its address has to be newly programmed. I have been testing on our meeting the layout equipped with the LNiWI interface LocoNet/WiThrottle enabling communication with the wifi throttle ESU mobile control II. When some slot was already occupied by the connected wired FRED, then when I wanted to choose the same address on the ESU throttle, it was not possible and the throttle informed me on the display that it is impossible. ESU throttle obviously checks if the address (the slot) is free. wiFRED should function similar way - at least letting know the user that the address is occupied and respective train can not be controlled. I hope I explained what I mean. 
Sorry for my bad English, I hope you understand what I mean.
Regards
Jan

newHeiko, 2023-05-26:
What the wiFred does when its address is already in use by another throttle depends entirely on how it's connected to Loconet / the DCC system.

JMRI will notify the wiFred if the address is already in use. The wiFred will not control the loco then (although it will not forget it's configuration, so the next time you turn it on, it will try again).

And I will implement some LED pattern to tell the user "that loco is busy". Soon. That's what #45 is about.

I don't know how the LNWI handles that case.

Heiko

pischky, 2023-05-28:
I do not really understand the request. Using a computer or mobile phone to setup wiFred to use a specific loco is part of the design and there is no need to provied another way. This is also a saftey feature required.
I vote to reject this request.