newHeiko/wiFred #51: Support of 28 Speed Steps

pischky, 2023-05-31:
wiThrottle should support DCC decoders with 28 speed steps.

There should be a selection after the DCC address with at least the options "128", "28" speed steps.
May be this could extended to all options available on a Loconet command station (14, 128 DAC, 28 DAC, SX, ...).

I assume that this requires an extension of the wiThrottle protocol, selection of speed steps in the wiThrottle server is much to complicated for the average user.

There are many (older) decoders (like Lenz LE103, LE040) that does not work with 128 speed steps but are still used. Also there are some decoders that support speed table only when used with 28 speed steps.

pischky, 2023-05-31:
> actually there is a command to set the speed step mode

I did not find this. Any Hint?

> I'll look into it, though I won't be able to test on anything else beside JMRI.

I will test it from throttle to loco. I have DCS100 and Loco with LE103 (or DCC Sniffer) for this.

newHeiko, 2023-05-31:
> Throttle Commands >> 's' set speed step mode.

:) I'd hope it's the same as these: https://www.jmri.org/help/en/package/jmri/jmrit/withrottle/Protocol.shtml#ThrottleChangeNotification

(though I don't know what 27 speed steps is)

pischky, 2023-05-31:
> though I don't know what 27 speed steps is

Older Dekoders from Lenz (LE075 or Roco 10741) supported only 14 and 27 speed steps. 27 was like 14, but the CU was sending two different speed steps alternating. [http://www.dcc-mueller.de/decoder/dectab_o.htm](http://www.dcc-mueller.de/decoder/dectab_o.htm)

pischky, 2023-06-01:
About guessing the values:

JMRI has a class (enum) jmri.SpeedStepMode. The ordinal values would be 0=UNKNOWN, 1=NMRA_DCC_128, 2=NMRA_DCC_28, 3=NMRA_DCC_27, 4=NMRA_DCC_14, 5=MOTOROLA_28, 6=TMCC_32, 7=INCREMENTAL.

That does not match what is documented in ThrottleChangeNotification.

https://www.jmri.org/JavaDoc/doc/src-html/jmri/SpeedStepMode.html
https://www.jmri.org/JavaDoc/doc/jmri/SpeedStepMode.html

May be a hint is the method decodeSpeedStepMode() in jmri.jmrit.withrottle.ThrottleController:
https://www.jmri.org/JavaDoc/doc/src-html/jmri/jmrit/withrottle/ThrottleController.html#line.734

So I would guess that possible values are

- "1", 
- "2"
- "4"
- "8"
- "16"

and 

- "unknown"
- "128"
- "28"
- "27"
- "14"
- "motorola_28"
- "tmcc_32"
- "incremental"