newHeiko/wiFred #52: Selection of long/short address should be removed

pischky, 2023-05-31:
Selection of long or short address should be removed in configuration panel.

On Loconet only short adresses are used for range 1..127. Long adresses are not supported in this range.

The option is confusing and can be done automatically from address (if (a<128) "short" else "long")

danielb987, 2023-05-31:
> The option is confusing and can be done automatically from address (if (a<128) "short" else "long")

No. Some DCC systems have (if (a<100) "short" else "long").

WiThrottle is not only used on LocoNet. It can be used on XpressNet as well, for example.

pischky, 2023-05-31:
> It can be used on XpressNet as well, for example.
I would never come up with the idea to use XpressNet. But technical you are right the wiThrottle protocol can be use there.

The question here is about usage of wiFRED.

danielb987, 2023-05-31:
> The question here is about usage of wiFRED.

There is nothing that says that WiFred will only be used on LocoNet. I wrote WiThrottle but I ment WiFred. I have a WiFred myself and I will probably use it on the club layout which uses XpressNet and Z21, as well as Fremo meetings with LocoNet.

pischky, 2023-05-31:
> I wrote WiThrottle but I ment WiFred

Ah. Ok I understand. 

Sorry I'm a little bit focused on use with FREMODCC (https://fremodcc.sourceforge.net/). And it would help a little bit to remove the long/short option. 

But you will loose 28 possible addresses of 10239 when used on XpressNet. Thats right.

newHeiko, 2023-05-31:
I won't remove that option, but let me think about how to change the behavior so Fremo users won't have to touch it...